aws/amazon-freertos #3534: [Feature Request] Update submodule "vendors/espressif/esp-idf" to release/v4.4 or esp-afr-sdk

energous-ameen, 2022-09-23:
**Describe the new feature request**
The esp-idf release/v4.4 or later supports external coex interface. This feature is required for the design with esp32s2 and external BLE chip that shares single antenna. 

**Describe other alternatives you've considered**
I couldn't find alternative. The coex function are part of esp_wifi componenent libnet80211.a

johnrhen, 2022-09-26:
Hello - thank you for submitting this feature request, I will bring it up with the team.

n9wxu, 2022-11-11:
The recommended path is through the ESP32c3 golden reference integration here: 
https://github.com/FreeRTOS/iot-reference-esp32c3
The C3 golden reference uses IDF so it will also build correctly for the S3.

closed 2022-11-11 by n9wxu