aws/amazon-freertos #3535: [General] Unable to access esp-idf ble_wifi_provisioning component

genie-ltran, 2022-09-27:
I'm attempting to us "ble_wifi_provisioning" component from the esp-idf library.  Under the top level CMakeLists.txt, I've included the component to be linked, yet it doesn't recognize the header files where app_main() is located.  Where do I need to look to makes the necessary changes for this to work.

I'm 

- Hardware board: ESP32-WROOM-32E
- IDE used: None
- Operating System: Windows 10
- Code version: 202012.01
- Project/Demo: mqtt_demo_mutual_auth

From CMakeLists.txt
`
cmake_minimum_required(VERSION 3.13)

project(freertos_examples)

add_executable(afr_demo src/main.c)

# Tell IDF build to link against this target.
set(IDF_PROJECT_EXECUTABLE afr_demo)

# Add some extra components. IDF_EXTRA_COMPONENT_DIRS is an variable used by ESP-IDF
# to collect extra components.
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/sys" ABSOLUTE
)
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})

get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/foo" ABSOLUTE
)
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})

# As of now there's no offical way to redefine config files outside of FreeRTOS source tree.
# This is a temporary approach to inject an include path so that this takes precedence over the
# config file directory inside FreeRTOS.
include_directories(BEFORE freertos-configs)

# Add freertos as an subdirectory. AFR_BOARD tells which board to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
add_subdirectory(freertos)

# Link against the mqtt demo so that we can use it. Dependencies of this demo are transitively
# linked.
target_link_libraries(
    afr_demo
    PRIVATE
    AFR::demo_core_mqtt
    AFR::common_io
    AFR::ble_wifi_provisioning
)
`

In main.c
`
#include "wifi_provisioning\manager.h"
#include "wifi_provisioning\wifi_config.h"
#include "wifi_provisioning\scheme_ble.h"
.
.
.
    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_PROV_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler, NULL));
`

Compile error
`
../src/main.c
../src/main.c:70:39: fatal error: wifi_provisioning\manager.h: No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`

kstribrnAmzn, 2022-09-29:
Taking a look. Getting my workspace setup for my mac. I'll reply when I have more information on your issue.

kstribrnAmzn, 2022-09-29:
Some questions for you:
1. How did you download the code? Did you check out aws/freertos from Github or did you use the AWS 'Device Software' page in the AWS console?
2. Have you recursively checked out submodules? I checked out the code from the repo, synced my submodules, and am seeing the wifi_provisioning files under _/vendors/espressif/esp-idf/components/wifi_provisioning/include/wifi_provisioning_
3. Have you tried building off of the existing BLE demo? You can select a demo following the instructions [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_esp32wroom-32se.html#configure-demos-esp32wroom-32s-idf42)

genie-ltran, 2022-10-03:
1.	How did you download the code? Did you check out aws/freertos from Github or did you use the AWS 'Device Software' page in the AWS console?

I checked it out it directly from the AWS github repo.

2.	Have you recursively checked out submodules? I checked out the code from the repo, synced my submodules, and am seeing the wifi_provisioning files under /vendors/espressif/esp-idf/components/wifi_provisioning/include/wifi_provisioning

It was recursively checked out.  I can see the source.  I just can’t reference certain files in the make process.

3.	Have you tried building off of the existing BLE demo? You can select a demo following the instructions [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_esp32wroom-32se.html#configure-demos-esp32wroom-32s-idf42)

Building individual demos works, but I need to combine features from multiple demos.  That’s when I have problems.

kstribrnAmzn, 2022-10-10:
Some further questions because I'm failing to set this up (or understand how it was setup).

You're directory structure looks something like the following, right?

```
|- CMakeLists.txt
|- src
|    |- main.c
|- freertos (amazon-freertos clone)
|    |-demos
|    |-freertos-kernel
|    |-(etc...)
```

Can you tell me more about how you've structured your project?

_______________

When I try this directory structure I'm greeted by 2 errors when I run `cmake -B build -S .`

This first error seems to stem from the fact that the `IDF_PROJECT_EXECUTABLE` is being added as an executable both in this top level CMakeLists.txt of yours and in the esp32 board CMakeLists.txt found [here](https://github.com/aws/amazon-freertos/blob/main/vendors/espressif/boards/esp32/CMakeLists.txt#L434-L437).
```
CMake Error at amazon-freertos/vendors/espressif/boards/esp32/CMakeLists.txt:434 (add_executable):
  add_executable cannot create target "afr_demo" because another target with
  the same name already exists.  The existing target is an executable created
  in source directory "/Users/kstribrn/workspace/oncall".  See documentation
  for policy CMP0002 for more details.
Call Stack (most recent call first):
  amazon-freertos/CMakeLists.txt:78 (include)
```

This second error might stem from the first one.
```
CMake Error at amazon-freertos/vendors/espressif/esp-idf/components/esp32/project_include.cmake:21 (message):
  Internal error, toolchain has not been set correctly by project (or an
  invalid CMakeCache.txt file has been generated somehow)
Call Stack (most recent call first):
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/build.cmake:312 (include)
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/build.cmake:457 (__build_process_project_includes)
  amazon-freertos/vendors/espressif/boards/esp32/CMakeLists.txt:595 (idf_build_process)
  amazon-freertos/CMakeLists.txt:78 (include)
```

genie-ltran, 2022-10-11:
1.  How did you download the code? Did you check out aws/freertos from Github or did you use the AWS 'Device Software' page in the AWS console?
I checked it out it directly from the AWS github repo.

  1.  Have you recursively checked out submodules? I checked out the code from the repo, synced my submodules, and am seeing the wifi_provisioning files under /vendors/espressif/esp-idf/components/wifi_provisioning/include/wifi_provisioning
It was recursively checked out.  I can see the source.  I just can’t reference certain files in the make process.

  1.  Have you tried building off of the existing BLE demo? You can select a demo following the instructions here<https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_esp32wroom-32se.html#configure-demos-esp32wroom-32s-idf42>
Building individual demos works, but I need to combine features from multiple demos.  That’s when I have problems.

genie-ltran, 2022-10-11:
I structured my project where the amazon-freertos repo is brought in a submodule.  The reason is keep my code separate from the amazon repo.  I structured it this way based on the ticket linked [here](https://github.com/aws/amazon-freertos/issues/2978).

I then added my individual components in its own directory with a CMakeLists.txt file to reference include and source directories.

kstribrnAmzn, 2022-11-01:
Thanks for linking to the [amazon-freertos-examples](https://github.com/tgsong/amazon-freertos-examples) repo. I dug around a bit and think I found what might be the source of your issue - though I don't have the solution just yet.

[amazon-freertos-examples](https://github.com/tgsong/amazon-freertos-examples) submodules [this commit](https://github.com/aws/amazon-freertos/tree/fd27d7bbd0d370004817c217b7fb95242f136b0d) of amazon-freertos which in turn submodules Espressif's esp-idf ([this commit](https://github.com/espressif/esp-afr-sdk/tree/29889b575e9f6f72aada23ab24d82d8d177498f0)).

This is important as it's the esp-idf 'wifi_provisioning' *component* which includes the wifi_provisioning/* header files your build is failing to find - see [the wifi_provisioning component Cmake](https://github.com/espressif/esp-afr-sdk/blob/29889b575e9f6f72aada23ab24d82d8d177498f0/components/wifi_provisioning/CMakeLists.txt). The AWS 'ble_wifi_provisioning' *module* from AFR does not include these headers (from what I'm seeing). The 'ble_wifi_provisioning' module comes from the c-sdk which is independent of platform - see [this Cmake](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/libraries/c_sdk/standard/ble/CMakeLists.txt).

As for how to fix this - I'm still figuring this out. The [esp-idf documentation](https://docs.espressif.com/projects/esp-idf/en/v3.3.5/api-guides/build-system-cmake.html) suggests that this component is available and included by default but this is clearly not the case.

____________
As for why I think the component is available, below are some snippets from building the [tgsong/amazon-freertos-examples](https://github.com/tgsong/amazon-freertos-examples)...

```
-- Building for target esp32
-- Component names: soc log heap freertos vfs newlib esp_ringbuf driver esp_event ethernet mbedtls micro-ecc efuse bootloader_support partition_table app_update spi_flash nvs_flash lwip tcpip_adapter xtensa-debug-module app_trace pthread smartconfig_ack wpa_supplicant espcoredump esp32 cxx foo amazon-freertos-common secure_sockets bootloader nimble bt console esp_adc_cal nghttp esp_http_server esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json mdns openssl protobuf-c protocomm spiffs ulp wifi_provisioning
-- Component paths: [some paths redacted for brevity];/Users/kstribrn/amazon-freertos-examples/freertos/vendors/espressif/esp-idf/components/wifi_provisioning
...
...
.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DAFR_ESP_LWIP -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H [some more removed stuff] -I/Users/kstribrn//amazon-freertos-examples/freertos/vendors/espressif/esp-idf/components/wifi_provisioning/include
```

kstribrnAmzn, 2022-11-01:
Played around some more and found a couple work arounds...

**The simplest** for me was to change your imports from using a backslash to a forward slash. This should fix your build. From what I found backslash support is a per-compiler thing. The xtensa compiler might just not handle them as you expect.

```
#include "wifi_provisioning/manager.h"
#include "wifi_provisioning/wifi_config.h"
#include "wifi_provisioning/scheme_ble.h"
```

Now the **more involved** thing you can try if this doesn't work is to copy `aws-freertos-examples/freertos/vendors/espressif/esp-idf/components/wifi_provisioning/include/wifi_provisioning/*` to a single level up (aka `aws-freertos-examples/freertos/vendors/espressif/esp-idf/components/wifi_provisioning/include/*`). This results in two copies of the header files. You'll also need to modify your imports in `main.c` to...

```
#include "manager.h"
#include "wifi_config.h"
//#include "scheme_ble.h" //commenting this out as the copy causes further path issues.
```
and build this. If this works but the forward slash didn't work then the issue could lie in the specific commit of esp-idf in Amazon FreeRTOS (though I'm doubtful of this).

kstribrnAmzn, 2022-11-16:
I'm going to close this since there hasn't been further activity. Feel free to reopen this if your issue still exists.

closed 2022-11-16 by kstribrnAmzn