aws/amazon-freertos #3536: [General] Cannot use Bluedroid, NVS crashes/panics

taherrera, 2022-09-29:
Hello,

I am trying to use bluetooth classic on the ESP32. For this I basically copied the Arduino implementation of bluetooth for the ESP32.
On ESP-IDF v4.4, this bluetooth implementation works perfectly, however on Amazon-Freertos it does not work.

To use this library, I have to change the menuconfig and use these configurations:
Bluetooth controller -> BR/EDR Only
Bluetooth Host -> Bluedroid - Dual-mode
Bluedroid options -> Classic Bluetooth
Bluedroid options -> Classic Bluetooth -> SPP

When I build the project, these lines stand out for me:

```
-- Project sdkconfig file /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/sdkconfig
Loading defaults file /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/sdkconfig.defaults...
/home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/sdkconfig.defaults:21 CONFIG_NIMBLE_ENABLED was replaced with CONFIG_BT_NIMBLE_ENABLED
/home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/sdkconfig.defaults:22 CONFIG_NIMBLE_MAX_CONNECTIONS was replaced with CONFIG_BT_NIMBLE_MAX_CONNECTIONS
...
...
**/tmp/confgen_tmpdfx8oq18:20 line was updated to CONFIG_BT_BLUEDROID_ENABLED=n**
...
```
Is Bluedroid being disabled by Amazon-FreeRTOS ? I am confused by this line.

When I initialize bluetooth I can see it broadcasting on my phone, however, when I try to pair I get this error:

```
Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x400d38d8  PS      : 0x00060734  A0      : 0x80083765  A1      : 0x3ffbe0f0  
0x400d38d8: xQueueReceiveFromISR at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/queue.c:1834

A2      : 0x3ff81928  A3      : 0x3ffbe110  A4      : 0x3ff101b4  A5      : 0x4008e104  
0x4008e104: _frxt_int_enter at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/portasm.S:119

A6      : 0x00000100  A7      : 0x3ffcaec0  A8      : 0x800826a4  A9      : 0x00000001  
A10     : 0x3ff81928  A11     : 0x00000000  A12     : 0x3ffbe110  A13     : 0x3ffb9274  
A14     : 0x40088d34  A15     : 0x3ffcaed1  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
0x40088d34: r_ld_fm_frame_isr at ??:?

EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core  0 was running in ISR context:
EPC1    : 0x4008d53a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d38d8
0x4008d53a: coex_is_in_isr_wrapper at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp_wifi/esp32/esp_adapter.c:577

0x400d38d8: xQueueReceiveFromISR at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/queue.c:1834

Backtrace:0x400d38d5:0x3ffbe0f0 0x40083762:0x3ffbe110 0x40083b7e:0x3ffbe150 0x4008370d:0x3ffbe190 0x40080db5:0x3ffbe1b0 0x400887f2:0x3ffbe1d0 0x40088d4a:0x3ffbe1f0 0x4008c866:0x3ffbe220 0x4008d1e7:0x3ffbe240 0x400813ee:0x3ffbe260 0x40082bba:0x3ffcaef0 0x40082ea4:0x3ffcaf10 0x40091115:0x3ffcaf40 0x400916dd:0x3ffcaf60 0x400834c6:0x3ffcaf90 0x40083539:0x3ffcafe0 0x400fe3ad:0x3ffcb000 0x400fd399:0x3ffcb020 0x400fd685:0x3ffcb040 0x400fc135:0x3ffcb0b0 0x400fc581:0x3ffcb130 0x400fcbb7:0x3ffcb1a0 0x400fbc57:0x3ffcb1c0 0x400f4756:0x3ffcb1f0 0x400eb59a:0x3ffcb250 0x400d5841:0x3ffcb270 0x400eb7ed:0x3ffcb2c0 0x400eba2a:0x3ffcb410 0x400e7ee9:0x3ffcb440 0x400e9026:0x3ffcb460
0x400d38d5: xQueueTakeMutexRecursive at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/queue.c:706

0x40083762: coex_bb_reset_unlock at ??:?

0x40083b7e: coex_core_release at ??:?

0x4008370d: coex_bt_release at ??:?

0x40080db5: coex_bt_release_wrapper at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/bt/controller/bt.c:1047

0x400887f2: coex_classic_bt_release$part$0 at ld_fm.c:?

0x40088d4a: r_ld_fm_frame_isr at ??:?

0x4008c866: r_rwbt_isr at ??:?

0x4008d1e7: r_rwbtdm_isr_wrapper at intc.c:?

0x400813ee: _xt_lowint1 at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1193
[... 975 lines cut ...]
esp_err_t BluetoothSerial_register_callback(esp_spp_cb_t * callback);

void BluetoothSerial_onConfirmRequest(ConfirmRequestCb cb);
void BluetoothSerial_onAuthComplete(AuthCompleteCb cb);
void BluetoothSerial_confirmReply(bool confirm);

void BluetoothSerial_enableSSP();
bool BluetoothSerial_setPin(const char *pin);
bool BluetoothSerial_connect_name(char * remoteName);
bool BluetoothSerial_connect_addr(uint8_t remoteAddress[]);
bool BluetoothSerial_connect();
bool BluetoothSerial_connected(int timeout/*=0*/);
bool BluetoothSerial_isReady(bool checkMaster/*=false*/, int timeout/*=0*/);
bool BluetoothSerial_disconnect();
bool BluetoothSerial_unpairDevice(uint8_t remoteAddress[]);

char * BluetoothSerial_local_name;

#else
#error You must enable bluetooth classic in menuconfig and Bluetooth controler mode (BR/EDR Only)

#endif

#endif
```

kstribrnAmzn, 2022-10-04:
I'm going to start looking into this. I hope to have an answer for you in the next few days.

kstribrnAmzn, 2022-10-05:
A few things I noticed...

>To use this library, I have to change the menuconfig and use these configurations:

_This_ library being the esp-idf library?
______________

>Is Bluedroid being disabled by Amazon-FreeRTOS ? I am confused by this line.

I don't think so. This configuration looks to be set in an sdk configuration file provided by esp-idf. See [this](https://github.com/espressif/esp-idf/blob/release/v4.2/examples/bluetooth/nimble/bleprph/sdkconfig.defaults#L11). You may need to set this value.
______________

Can you provide a link to whatever you're basing your demo off of? Looking through your code I see where you create the Bluetooth task but I'm not seeing the scheduler being started anywhere. Is this being started somewhere else?

______________

Near the very top of your stack trace points to [this function](https://github.com/espressif/esp-idf/blob/release/v4.2/components/bt/common/btc/core/btc_task.c#L184) in esp-idf. From there it spirals into the esp-idf framework a bit. Your post may be better answered by the Espressif/ESP32 forums as this might be as simple as a configuration issue.

taherrera, 2022-10-06:
Hello,

> This library being the esp-idf library?

I mean the library I translated from [here](https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial) into my project.

> I don't think so. This configuration looks to be set in an sdk configuration file provided by esp-idf. See [this](https://github.com/espressif/esp-idf/blob/release/v4.2/examples/bluetooth/nimble/bleprph/sdkconfig.defaults#L11). You may need to set this value.

I changed the configurations on the file you are mentioning to:

```
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=n
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=y
CONFIG_BTDM_CTRL_MODE_BTDM=n
CONFIG_BT_BLUEDROID_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=n
```
But I still see the line:
`/tmp/confgen_tmptfuyjhj6:20 line was updated to CONFIG_BT_BLUEDROID_ENABLED=n`

> Can you provide a link to whatever you're basing your demo off of? Looking through your code I see where you create the Bluetooth task but I'm not seeing the scheduler being started anywhere. Is this being started somewhere else?

I am not sure how to start the scheduler, you mean `vTaskStartScheduler()`? I was not calling this function, so I added this to the main.c but it still fails, now with a slightly different error though. I uploaded the entire project to: [https://github.com/taherrera/AFR-Bluedroid](https://github.com/taherrera/AFR-Bluedroid). This is the error:

```
Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x400d3754  PS      : 0x00060034  A0      : 0x8008cf46  A1      : 0x3ffbe1f0  
0x400d3754: xQueueGenericSendFromISR at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/queue.c:987

A2      : 0x3ffbc404  A3      : 0x3ffbe210  A4      : 0x3ffbe218  A5      : 0x3ffce250  
A6      : 0x3ffbe68c  A7      : 0x3ffd37fc  A8      : 0x80080e25  A9      : 0x3ffbe1e0  
A10     : 0x3ffbc404  A11     : 0x3ffbe210  A12     : 0x3ffbe218  A13     : 0x00000000  
A14     : 0xfffff000  A15     : 0x00000100  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core  0 was running in ISR context:
EPC1    : 0x40008544  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d3754
0x400d3754: xQueueGenericSendFromISR at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/queue.c:987

Backtrace:0x400d3751:0x3ffbe1f0 0x4008cf43:0x3ffbe210 0x4008d1ed:0x3ffbe240 0x400813ee:0x3ffbe260 0x40008541:0x3ffce180 0x400835ad:0x3ffce1a0 0x40091523:0x3ffce1c0 0x40091436:0x3ffce200 0x40091725:0x3ffce220 0x400834c6:0x3ffce250 0x40083539:0x3ffce2a0 0x401224c5:0x3ffce2c0 0x40121689:0x3ffce2e0 0x40121809:0x3ffce300 0x401201b6:0x3ffce370 0x401206f9:0x3ffce3f0 0x40120d2f:0x3ffce460 0x4011fe73:0x3ffce480 0x401007f2:0x3ffce4b0 0x400d6626:0x3ffce510 0x400d678d:0x3ffce530 0x40105692:0x3ffce580 0x400ff721:0x3ffce740 0x40101607:0x3ffce760
0x400d3751: prvInitialiseMutex at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/queue.c:509
 (inlined by) prvInitialiseMutex at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/queue.c:492
 (inlined by) xQueueCreateMutex at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/queue.c:528

0x4008cf43: btdm_task_post_from_isr at ??:?

0x4008d1ed: r_rwbtdm_isr_wrapper at intc.c:?

0x400813ee: _xt_lowint1 at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1193

0x400835ad: delay_us at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_os_func_app.c:102

0x40091523: spi_flash_chip_generic_wait_idle at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_chip_generic.c:283

0x40091436: spi_flash_chip_generic_page_program at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_chip_generic.c:175
 (inlined by) spi_flash_chip_generic_page_program at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_chip_generic.c:165

0x40091725: spi_flash_chip_generic_write at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_chip_generic.c:195

0x400834c6: esp_flash_write at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/esp_flash_api.c:616

0x40083539: spi_flash_write at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/esp_flash_api.c:756

0x401224c5: nvs::nvs_flash_write(unsigned int, void const*, unsigned int) at /home/tom/Documents/SLT/tecnocal/instacrop/fw/instacrop_test/build/../amazon-freertos/vendors/espressif/esp-idf/components/nvs_flash/src/nvs_ops.cpp:72

[... cut ...]
I will ask, but it is strange because this works on ESP-IDF, so there is something different going on with AFR, causing this error. The only difference in the implementation on ESP-IDF are these functions on my main.c which I do not know why I need them on AFR, but without them I cannot compile Amazon-FreeRTOS:

```
extern void esp_vApplicationTickHook();
void IRAM_ATTR vApplicationTickHook(){
    esp_vApplicationTickHook();
}

extern void esp_vApplicationIdleHook();
void vApplicationIdleHook(){
    esp_vApplicationIdleHook();
}

void vApplicationDaemonTaskStartupHook( void ){
}
```

Skptak, 2022-10-10:
Hey, I've cloned down your repository and have started looking into your problem. I'll continue this effort to see if I can better assist you with your issue.

In the meantime I was wondering if you had tried starting with an existing bluetooth demo for the STM32 board and then from there adding your needed functionality? This could ensure that you have the board and device properly working by following our documentation, at which point you could begin adding in your functions for your personal use case? We have a doc for our bluetooth demos available [here](https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html), as well as a GitHub link to some demos [here](https://github.com/aws/amazon-freertos/tree/main/demos/ble).

If any of these resources help you solve your problem please let me know!

taherrera, 2022-10-11:
Hello @laukik-hase ,

I tried PR #3541  , with these commands

```
git clone https://github.com/aws/amazon-freertos.git --recurse-submodules
cd amazon-freertos
git fetch origin pull/3541/head:afr_branch
git checkout afr_branch
git submodule update --init --recursive --checkout
```

But it fails on `git submodule update --init --recursive --checkout` with this error:

```
fatal: reference is not a tree: 3ae7e4abf19a00196106d6aa4cdda27ce40cf165
fatal: reference is not a tree: b967ca054f48a36f82d8fcdd32e54ec5144f2751
Unable to checkout 'b967ca054f48a36f82d8fcdd32e54ec5144f2751' in submodule path 'vendors/espressif/esp-idf/components/nghttp/nghttp2/third-party/neverbleed'
Failed to recurse into submodule path 'vendors/espressif/esp-idf/components/nghttp/nghttp2'
Unable to checkout '3ae7e4abf19a00196106d6aa4cdda27ce40cf165' in submodule path 'freertos_kernel'
Failed to recurse into submodule path 'vendors/espressif/esp-idf'
```

**However, I tried ESP-IDF checkout release/v4.2 and I can pair successfully, so it works con ESP-IDF 4.2 and 4.4.** Now I am pretty sure there is something not allowing me to run this code with Amazon-FreeRTOS and it is not an issue originating in ESP-IDF.

Hello @Skptak,

> In the meantime I was wondering if you had tried starting with an existing bluetooth demo for the STM32 board and then from there adding your needed functionality? This could ensure that you have [...]

I have not tried other bluetooth solutions, because *I think* there is an underlying issue behind this that causes us not only not be able to use this bluetooth code, but also we have not been able to use NVS storage and cannot use the [ESP-IDF 4.2 OTA functions](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/api-reference/system/ota.html). So since there are 3 problems related somehow to the flash storage in Amazon-FreeRTOS, we have not tried it. I first published this NVS problem as a bug but it was rapidly dismissed (see #3531) . So I lifted this issue with this bluetooth code which is more reproducible.

Skptak, 2022-10-12:
Hey again,

>  I have not tried other bluetooth solutions, because I think there is an underlying issue behind this that causes us not only not be able to use this bluetooth code
Sorry, my intention was not to tell you to use a different bluetooth solution. I was trying to offer that by starting with one of our bluetooth demos you could verify that everything in your build system and board is working correctly. For example if you take a look at this [app_main() function](https://github.com/aws/amazon-freertos/blob/main/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L134) from the demos. You can see that there are some additional checks that happen, and functions that are called when performing the BLE setup. I felt that it might relevant to look over some of this since you're running into several different issues, directly related to using the board.

Additionally, I can't speak to the error you're receiving on the update to that pull request. I was able to do the update to that PR and then perform the build. I don't currently have access to an ESP32 to run your code to further assist you, but I'm in the process of getting one so I can help you further.

If you have any updates while I work on getting a board, please let me know.

dachalco, 2022-10-21:
@taherrera 

To reaffirm, you changed the esp32 configs as such?
```
Bluetooth controller -> BR/EDR Only
Bluetooth Host -> Bluedroid - Dual-mode
Bluedroid options -> Classic Bluetooth
Bluedroid options -> Classic Bluetooth -> SPP
```

If so then the the Bluetooth controller is configured for Classic Bluetooth only, but the amazon-freertos bluetooth services are GATT services that _require_ BLE. Can you try setting Bluetooth controller to Bluetooth Dual Mode instead? You should be able to use Bluedroid, but building with amazon-freertos, NimBLE is set as the default.

Also amazon-freertos BLE services also directly use the bluedroid/nimBLE APIs. There may be some collision/clobbering, for example amazon-freertos bluetooth middleware also configures GAP.

Additionally, to clarify amazon-freertos does not technically use _any_ version of ESP-IDF, rather it uses a fork specifically maintained for amazon-freertos, by Espressif: [espressif/esp-afr-sdk](https://github.com/espressif/esp-afr-sdk).

@mahavirj May have additional info to add.

taherrera, 2022-10-21:
Thanks @dachalco , I will try your suggested change and inform you asap.

Also, I forgot to mention, that when I *do not* initialize the nvs, the bluetooth implementation works (although I have to pair each time to the device before connecting).

dachalco, 2022-10-21:
Having to pair every time, due to omitting NVS, is expected. The bond information is supposed to persist, but if it doesn't then devices need to negotiate and exchange key information again, since it wasn't saved.

taherrera, 2022-10-21:
The thing I am trying to point out is that Bluetooth works as expected and does not panic if I do not initialize the nvs.

dachalco, 2022-10-21:
How are you building this code in amazon-freertos? From what I understand you've modifed the `aws_demos/application_code/main.c` for the esp32? What's the full diff of changes made to `main.c`? As mentioned earlier there's setup being done there for BLE and related NVS that should be retained. Furthermore the Arduino API should be reviewed to ensure it doesn't clobber configurations made by amazon-freertos demo.

Additionally, as I understand your code was based on IDF v4.4, but as mentioned amazon-freertos uses a fork based on v4.2. Have you reviewed that functions and usage are the same for both?

taherrera, 2022-10-24:
I set this in menuconfig:
` Component config → Bluetooth → Bluetooth controller → Bluetooth controller mode (BR/EDR/BLE/DUALMODE)` but now this implementation cannot be compiled due missing reference to this function: `btsnd_hcic_ble_update_adv_report_flow_control`

The application that throws these errors is here: [https://github.com/taherrera/AFR-Bluedroid)](https://github.com/taherrera/AFR-Bluedroid)

n9wxu, 2022-10-27:
In the AFR demos, there are a number of conflicts with a bluedroid SPP configuration.  These conflicts are in iot_ble_hal_common_gap.c.  This file configures the bluetooth for BLE GAP and only does incomplete for bluedroid intialization if the bluedroid menu option is selected.  Compiling with the IDF 4.2 SPP demo incorporated is possible but a lot of work is required to get SPP functioning.

n9wxu, 2022-11-02:
The Amazon FreeRTOS product does not support bluetooth classic.  It does support bluedroid in BLE mode.

closed 2022-11-02 by n9wxu