aws/amazon-freertos #3539: ESP32 compiled binary shows absolute file path when walked through using binwalk

vineet288818, 2022-10-04:
We are evaluating the binary generated for ESP32.
The binary when run through binwalk shows absolute file paths for the files.

ENVIRONMENT :  AFR version : 202012

**Here is a sample of the file paths seen

264372        0x408B4         Unix path: /home/ubuntu/ic/code/freertos/freertos_kernel/stream_buffer.c
266912        0x412A0         Unix path: /home/ubuntu/ic/code/freertos/vendors/espressif/esp-idf/components/soc/esp32/rtc_time.c
268392        0x41868         Unix path: /home/ubuntu/ic/code/freertos/vendors/espressif/esp-idf/components/soc/src/memory_layout_utils.c
268844        0x41A2C         Unix path: /home/ubuntu/ic/code/freertos/vendors/espressif/esp-idf/components/vfs/vfs.c
270360        0x42018         Unix path: /home/ubuntu/ic/code/freertos/vendors/espressif/esp-idf/components/efuse/src/esp_efuse_utility.c
271032        0x422B8         Unix path: /home/ubuntu/ic/code/freertos/vendors/espressif/esp-idf/components/app_update/esp_ota_ops.c
272144        0x42710         Unix path: /home/ubuntu/ic/code/freertos/vendors/espressif/esp-idf/components/spi_flash/flash_mmap.c
272420        0x42824         Unix path: /home/ubuntu/ic/code/freertos/vendors/espressif/esp-idf/components/spi_flash/partition.c
278932        0x44194         Unix path: /home/ubuntu/ic/code/freertos/vendors/espressif/esp-idf/components/lwip/lwip/src/core/tcp_in.c
279948        0x4458C         Unix path: /home/ubuntu/ic/code/freertos/vendors/espressif/esp-idf/components/lwip/lwip/src/core/tcp_out.c**

How can we turn off inclusion of absolute file paths in the binary.

mahavirj, 2022-10-07:
@vineet288818 

Can you please try enabling following config options:

- `CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE` in menuconfig
- `CONFIG_FREERTOS_ASSERT_DISABLE` in menuconfig

Please see previous discussion on this topic at https://github.com/espressif/esp-idf/issues/6306 and https://github.com/espressif/esp-idf/issues/5873

dachalco, 2022-10-10:
Hi @vineet288818 Additionally you will need to remove the usage if `__FILE__` in `logging_stack.h` as such:
```
--- a/libraries/logging/include/logging_stack.h
+++ b/libraries/logging/include/logging_stack.h
@@ -42,11 +42,8 @@
 
 /* Macro to extract only the file name from file path to use for metadata in
  * log messages. */
-#ifdef _MSC_VER
-    #define FILENAME           ( strrchr( __FILE__, '\\' ) ? strrchr( __FILE__, '\\' ) + 1 : __FILE__ )
-#else
-    #define FILENAME           ( strrchr( __FILE__, '/' ) ? strrchr( __FILE__, '/' ) + 1 : __FILE__ )
-#endif
+#define FILENAME           "<redacted>"
+
```

You may do a similar redaction with the line number in the same file.

Skptak, 2022-10-11:
Hey, @EI-viren-moradiya thanks for bringing this issue up. I was wondering if you could possibly supply your build command, the version of binwalk you're using, and then the binwalk command you are using to see this output. I believe this information would help myself and others in assisting you with this problem.
Thanks in advance!

EI-viren-moradiya, 2022-10-12:
Hi @Skptak,

Below is the build command which I am using to generate the firmware binary.
---------------------------------------------------------------------------------------------------
#### Generate Build files
---------------------------------------------------------------------------------------------------
---------------------------------------------------------------------------------------------------
#### Build the project
**make all**
---------------------------------------------------------------------------------------------------
---------------------------------------------------------------------------------------------------
**Binwalk version**
$ binwalk --help
**Binwalk v2.1.1**
Craig Heffner, http://www.binwalk.org
---------------------------------------------------------------------------------------------------
---------------------------------------------------------------------------------------------------
Command to see the output:
**binwalk ic.bin**
---------------------------------------------------------------------------------------------------

dachalco, 2022-10-13:
@EI-viren-moradiya 

The linked IDF issues make some mention of it, but the config doesn't cover all uses of `__FILE__` as some software is submoduled and not owned by IDF.

Could you do a diff of the binwalk output before and after setting those sdkconfigs. I'm noticing that there are still issues with various other stack's assertions that still use `__FILE__`. For example `LWIP_ASSERT`. Could you redefine `LWIP_ASSERT` to not use `__FILE__` and confirm it removes the LWIP files at least? Removing the other full paths will be similar.