aws/amazon-freertos #3540: [General] STM32L4 discovary board AWS IoT Tera Term Error

Obinex, 2022-10-07:
**Briefly summarize the issue being raised**
So I went through all the steps to connect the STM32L4 to the AWS Cloud although I have encountered an error with the MQTT (the error I am receiving on Tera Term is shown in the references below), how do I rectify this issue?

**Describe the desired outcome**
This is what the outcome should look like:
[image]

**System information**
- Hardware board: [STM32L4 Discovery Kit Board]
- IDE used: [STM32CubeIDE]
- Operating System: [ Windows ]
- Code version: (run ``git describe --tags`` to find it)
- Project/Demo: [ mqtt_demo_mutual_auth | shadow_demo | custom application | etc... ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code/Steps to reproduce the current status"`.

**References**
[image]

**Code/Steps to reproduce the current status**
If applicable, provide concrete instructions or the relevant code snippet(s) for reproducing the current status. Code should be wrapped in the ``cpp`` tag in order to be displayed clearly. For example:
```cpp
printf("Hello World")
```
**Additional context**
Add any remaining information not mentioned in the previous responses.

archigup, 2022-10-14:
What version of amazon-freertos are you using?

From the logs it seems that it is failing due to the incoming packet having type 1, which is a CONNECT packet; this should not be received from the server.

n9wxu, 2022-10-28:
@Obinex Any answers to the questions above?

AniruddhaKanhere, 2022-10-28:
I agree with @archigup (with a minor correction) - I don't think that packet of type '1' is valid at all. CONNECT packet is `0x10` when printed in the logs since the 1 is shifted 4 times into the higher nibble of the byte. 

Can you capture some logs using Wireshark? It will help us figure out the bug much faster.

AniruddhaKanhere, 2022-11-03:
@Obinex I shall be closing this issue now as we do not have information to work on.
If you want, feel free to reopen this issue or create a new one if you have more questions and/or concerns.

closed 2022-11-03 by AniruddhaKanhere