aws/amazon-freertos #3544: [General] Ethernet AWS MQTT DNS Network Error

gsbwallace1996, 2022-11-02:
**Briefly summarize the issue being raised**
MQTT is unable to connection to AWS endpoint with error occurring at DNS lookup. Similar to issue [#936](https://github.com/aws/amazon-freertos/issues/936). Error does not occur with the same endpoint link for connection over WiFi.

**Describe the desired outcome**
Successful MQTT Ethernet connection to AWS endpoint without DNS error. Below shows the intended result using a WiFi module instead of connection over Ethernet:
```cpp
[AWS WIFI] WINC1500 WIFI: DNS lookup:
Host:       a1c29udxup0qn9-ats.iot.eu-west-1.amazonaws.com
IP Address: 52.213.52.29
[AWS] [INFO ][MQTT][lu] Establishing new MQTT connection.
[AWS] [INFO ][MQTT][lu] (MQTT connection 0x200068e8, CONNECT operation 0x20009bf0) Waiting for operation completion.
[AWS] [INFO ][MQTT][lu] (MQTT connection 0x200068e8, CONNECT operation 0x20009bf0) Wait complete with result SUCCESS.
[AWS] [INFO ][MQTT][lu] New MQTT connection 0x24001ec0 established.
```

**System information**
- Hardware board: STM32H743
- IDE used: STM32CubeIDE
- Operating System: Windows
- Code version: FreeRTOS Kernel V10.2.0
- Project/Demo: iot_network_afr.c (Amazon FreeRTOS Platform V1.0.0)

**References**
The below DNS error debug log repeats indefinitely:

```cpp
[AWS] [ERROR][NET][lu] Failed to resolve a1c29udxup0qn9-ats.iot.eu-west-1.amazon
aws.com.                                                 
[AWS] [ERROR][MQTT][lu] Failed to establish new MQTT connection, error NETWORK E
RROR.                                           
[AWS] [ERROR][DEMO][lu] MQTT CONNECT returned error NETWORK ERROR.
[AWS] [INFO ][Shadow][lu] Shadow library cleanup done.
[AWS] [INFO ][MQTT][lu] MQTT library cleanup done. 
[AWS] [ERROR][DEMO][lu] Error running demo.                 
[IDLE] 133672 104312                                          
[ADMIN] Starting AWS Task                                        
[AWS] [INFO ][DEMO][lu] Successfully initialised the demo. Network type for the 
demo: 4                                                        
[AWS] [INFO ][MQTT][lu] MQTT library successfully initialized.
[AWS] [INFO ][Shadow][lu] Shadow library successfully initialized.
```

Despite attempts at modifying the link to produce a different result the same DNS server address is always given:
[image]

**Code to reproduce bug**
The following is where the code throws the DNS error within the iot_network_afr.c file:
```cpp
/* Clear the connection information. */
    ( void ) memset( pNewNetworkConnection, 0x00, sizeof( _networkConnection_t ) );

    /* Create a new TCP socket. */
    tcpSocket = SOCKETS_Socket( SOCKETS_AF_INET,
                                SOCKETS_SOCK_STREAM,
                                SOCKETS_IPPROTO_TCP );

    if( tcpSocket == SOCKETS_INVALID_SOCKET )
    {
        IotLogError( "Failed to create new socket." );
        IOT_SET_AND_GOTO_CLEANUP( IOT_NETWORK_SYSTEM_ERROR );
    }

    /* Set up connection encryption if credentials are provided. */
    if( pAfrCredentials != NULL )
    {
        status = _tlsSetup( pAfrCredentials, tcpSocket, pServerInfo->pHostName, hostnameLength );

        if( status != IOT_NETWORK_SUCCESS )
        {
            IOT_GOTO_CLEANUP();
        }
    }

    /* Establish connection. */
    serverAddress.ucSocketDomain = SOCKETS_AF_INET;
    serverAddress.usPort = SOCKETS_htons( pServerInfo->port );
    serverAddress.ulAddress = SOCKETS_GetHostByName( pServerInfo->pHostName );

    /* Check for errors from DNS lookup. */
    if( serverAddress.ulAddress == 0 )
    {
        IotLogError( "Failed to resolve %s.", pServerInfo->pHostName );
        IOT_SET_AND_GOTO_CLEANUP( IOT_NETWORK_SYSTEM_ERROR );
    }
```

**Steps already taken to overcome issue**

1. As noted in issue 936, modifying values in the FreeRTOSIPConfig.h can resolve the issue but this has not happened. The values I have set below:

```cpp
#define ipconfigSOCK_DEFAULT_RECEIVE_BLOCK_TIME    pdMS_TO_TICKS( 5000 )
#define ipconfigSOCK_DEFAULT_SEND_BLOCK_TIME       pdMS_TO_TICKS( 5000 )

#define ipconfigUSE_DNS_CACHE                      ( 1 )
#define ipconfigDNS_REQUEST_ATTEMPTS               ( 3 ) // Changed from 2
```

2. I have used Powershell nslookup for the ethernet connection and it returns:
```ps
PS C:\Users\User> nslookup a1c29udxup0qn9-ats.iot.eu-west-1.amazonaws.com
Server:  UnKnown
Address:  192.168.0.1

Non-authoritative answer:
Name:    a1c29udxup0qn9-ats.iot.eu-west-1.amazonaws.com
Addresses:  2a01:578:3::364b:e990
          2a01:578:3::364d:5172
          2a01:578:3::34d0:aa56
          2a01:578:3::3f8:197d
          2a01:578:3::22f1:60c6
          2a01:578:3::22f2:52cd
          2a01:578:3::341f:fa23
          2a01:578:3::3648:bd94
          52.213.144.6
          54.220.53.255
          54.154.138.253
          34.250.166.247
          52.31.106.148
          34.251.121.26
          52.31.195.63
          52.210.5.189
```

3. Tried modifying the link, removing -ats etc. but no change to DNS server address or response
4. Directly connected to 2 separate routers bypassing switch
5. Ensured WiFi build and Ethernet build used the same link variable
6. Checked MQTT broker port was set to 8883

Is there potentially a config setting I am not configuring correctly? I feel as though if there was a router issue the WiFi MQTT would also fail.

Any help is much appreciated,
Greg

AniruddhaKanhere, 2022-11-03:
Hello @gsbwallace1996 can you capture a Wireshark log?
I am curious to know what is the value of `ipconfigDNS_CACHE_NAME_LENGTH`.

I am assuming that you are using FreeRTOS+TCP. Also, can you try doing a DNS lookup using `SOCKETS_gethostbyname( "www.github.com" )` and see whether that succeeds?

Thanks

htibosch, 2022-11-06:
@AniruddhaKanhere wrote:

> Hello @gsbwallace1996 can you capture a Wireshark log?

I would have asked for a PACP as well. Can you attach it in a ZIP file?

> I am curious to know what is the value of ipconfigDNS_CACHE_NAME_LENGTH.

That is the size of the largest URL that the DNS software can handle. By default, it is 254 bytes, but your copy of FreeeRTOSIPConfig.h might define it smaller.

Note that a copy of the URL will be declared on-stack:
~~~c
    char pcName[ ipconfigDNS_CACHE_NAME_LENGTH ];
~~~
... which may cost 256/4 words of stack.

Also, while debugging, it would be interesting to see the logging from FreeeRTOS+TCP: see `ipconfigHAS_PRINTF` and `ipconfigHAS_DEBUG_PRINTF` on [FreeRTOS.org](https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/TCP_IP_Configuration.html).

Of course it is possible the run a test without DNS, by filling in an IP-address:
~~~diff
-    serverAddress.ulAddress = SOCKETS_GetHostByName( pServerInfo->pHostName );
+    serverAddress.ulAddress = FreeRTOS_inet_addr_quick( 52, 213, 52, 29 );
~~~

`FreeRTOS_inet_addr_quick()` returns a network-endian IP-address, no worries about the endianness.

gsbwallace1996, 2022-11-06:
Hi @htibosch @AniruddhaKanhere,

Apologies in advance for the long post, TLDR: I actually believe this issue to now be a potential DHCP issue or general connection issue. I've put below the steps I've taken to reach that conclusion.

I can confirm that the ipconfigDNS_CACHE_NAME_LENGTH is set to 254 and I am using FreeRTOS+TCP.

Like you suggested @htibosch I done print line debugging of the lower level functions to ensure everything was passed and triggering correctly. Starting with the cache:

```c
“03/11/2022 12:47:34”,[AWS] [ERROR][NET][lu] Failed to resolve [a1c29udxup0qn9-ats.iot.eu-west-1.amazonaws.com](http://a1c29udxup0qn9-ats.iot.eu-west-1.amazonaws.com/).
“03/11/2022 12:47:34”,[AWS] [ERROR][MQTT][lu] Failed to establish new MQTT connection, error NETWORK ERROR.
“03/11/2022 12:47:34”,[AWS] [ERROR][DEMO][lu] MQTT CONNECT returned error NETWORK ERROR.
“03/11/2022 12:47:34”,[AWS] [INFO ][Shadow][lu] Shadow library cleanup done.
“03/11/2022 12:47:34”,[AWS] [INFO ][MQTT][lu] MQTT library cleanup done.
“03/11/2022 12:47:34”,[AWS] [ERROR][DEMO][lu] Error running demo.
“03/11/2022 12:47:36”,[ADMIN] Starting AWS Task
“03/11/2022 12:47:36”,[AWS] [INFO ][DEMO][lu] Successfully initialised the demo. Network type for the demo: 4
“03/11/2022 12:47:36”,[AWS] [INFO ][MQTT][lu] MQTT library successfully initialized.
“03/11/2022 12:47:36”,[AWS] [INFO ][Shadow][lu] Shadow library successfully initialized.
“03/11/2022 12:47:36”,[AWS] SOCKETS_GetHostByName
“03/11/2022 12:47:36”,[AWS] FreeRTOS_gethostbyname
“03/11/2022 12:47:36”,[AWS] FreeRTOS_gethostbyname_a start
“03/11/2022 12:47:36”,[AWS] FreeRTOS_gethostbyname_a ipconfigUSE_DNS_CACHE
“03/11/2022 12:47:36”,[AWS] FreeRTOS_dnslookup pcHostName: [a1c29udxup0qn9-ats.iot.eu-west-1.amazonaws.com](http://a1c29udxup0qn9-ats.iot.eu-west-1.amazonaws.com/) (string not truncated when passed)
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:36”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:36”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[IDLE] 108296 104016
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0
“03/11/2022 12:47:37”,[AWS] xDNSCache[x].pcName: 0
“03/11/2022 12:47:37”,[AWS] IF xDNSCache[ x ].pcName[ 0 ] == 0 (cycles for 20 per ipconfigDNS_CACHE_ENTRIES)
“03/11/2022 12:47:37”,[AWS] xFound == pdFALSE
“03/11/2022 12:47:37”,[AWS] xLookUp != pdFALSE
“03/11/2022 12:47:37”,[AWS] FreeRTOS_dnslookup ulIPAddress: 0 (zero returned addresses from cache)
```

[... 209 lines cut ...]
Turning the DHCP off in the config hasn't yielded any result changes from using the static IP. Tried to see if the DHCP was retrieving any bytes from within the prvProcessDHCPReplies function:

```c
	lBytes = FreeRTOS_recvfrom( xDHCPData.xDHCPSocket, ( void * ) &pucUDPPayload, 0ul, FREERTOS_ZERO_COPY, &xClient, &xClientLength );
	configPRINTF( ( "lBytes: %d\r\n", lBytes ) );
	if( lBytes > 0 )
	{
		/* Map a DHCP structure onto the received data. */
		pxDHCPMessage = ( DHCPMessage_t * ) ( pucUDPPayload );
		vTaskDelay(100);
		configPRINTF( ( "if( lBytes > 0 )\r\n" ) );
...
```

But no luck as return corresponds to If no bytes are received before the configured block time expires then -pdFREERTOS_ERRNO_EWOULDBLOCK (-11) is returned.

Using Wireshark in combination with a switch or directly into my laptop I attempted to see anything coming from the device. I have yet been unable to see anything for the devices static IP. I came across this [link](https://forums.freertos.org/t/freertos-tcp-ip/11468) which suggests that the address 0.0.0.0 is indeed from the embedded trying to get an address in the direct connection but I not sure. 

I have attached some of the wireshark files that I took and the config file.
[dev_ETH_DIRECT.zip](https://github.com/aws/amazon-freertos/files/9946376/dev_ETH_DIRECT.zip)

Next steps are probably to alter the code to fire down UDP to establish data packets coming out.

Many thanks,
Greg

htibosch, 2022-11-07:
After you turned of your DHCP, I would expect to see packets from 192.168.0.105, but I don't see any in any of the PCAP files?

What I don't understand is the logging `dnsDNS_PORT: 4`, shouldn't that be either 0x3500 or 0x0035 (depending on the endianness) ?

> Hardware board: STM32H743

Note that [this](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/main/source/portable/NetworkInterface/STM32Hxx) is one of the latest ports, and you might have some start-up problems.

The most difficult things is to find the proper memory that is not cached, and that can be passed to DMA.
Please have a look at the [readme.md](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/portable/NetworkInterface/STM32Hxx/readme.md)

Have you tried the more basic things like:

- Define a static address and try ping
- Use DHCP and see if the works

My feeling is that not a packet reaches the wire.

> WINC1500 WIFI

Just curious, are you using this in combination with a STM32H7x?