aws/amazon-freertos #3545: [General] Is there a reason, why there is almost two months no merge in main?

thomasklaush, 2022-11-03:


htibosch, 2022-11-03:
I am not sure what you are asking for?

Looking at the [PR-list](https://github.com/FreeRTOS/FreeRTOS/pulls?q=is%3Apr), the last merge into main was done [here](https://github.com/FreeRTOS/FreeRTOS/pull/861), two weeks ago.

eelaix, 2022-11-03:
Is there a reason, why there is almost two months no merge in main?
on aws/amazon-freertos ?

AniruddhaKanhere, 2022-11-03:
Hello @thomasklaush and @eelaix,
Thank you for reaching out to us.

The amazon-freertos repository got too large and too complex for a simple and efficient experience. Thus, for a better user experience we adopted an alternative of having one demo per repo rather than all in one repo.

To that end, we have focused all our attention on the [FreeRTOS/FreeRTOS](https://github.com/FreeRTOS/FreeRTOS) repository and the new [Long Term Support](https://github.com/FreeRTOS/FreeRTOS-LTS) core libraries.
We also have reference integration repositories focusing in vendor specific devices. See the [list](https://github.com/FreeRTOS?q=iot-reference).

I would take a look at the above listed latest LTS release and the featured reference integrations for the best examples.

Let us know if that makes sense. Thanks!

AniruddhaKanhere, 2022-11-03:
I shall close this issue for now.
If you have any more questions/concerns or feedback, please feel free to re-open the issue or create a new one.

Thanks again,
Aniruddha

closed 2022-11-03 by AniruddhaKanhere

thomasklaush, 2022-11-04:
@AniruddhaKanhere this means, this repo (aws/amazon-freertos) will not be supported any more.  

So the Getting started guide from aws is not correct any more, since it's referencing on this repo
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_esp32wroom-32se.html  

If I got it correct, there will be a repo for each vendor or each board in the future?