aws/amazon-freertos #3548: [General] esp_ota_begin fails in simple example

diegoacuna, 2022-11-07:
Hi, I'm trying to use OTA firmware updates with Amazon FreeRTOS but so far I haven't been able to get it work. When using the esp_ota_begin function, the platform crashes.

**System information**
- Hardware board: Hardware board: ESP32-WROVER and ESP32-WROOM
- Operating System: Ubuntu 20.04
- Code version: 202107.00-75-g8710a1b23
- Project/Demo: custom

**Code/Steps to reproduce the current status**

[Here you can find a sample application](https://github.com/diegoacuna/afr-test-ota) with only the code to run a esp_ota_begin in the board, for reference these are the relevant lines:

```cpp
const esp_partition_t *next_update_partition = esp_ota_get_next_update_partition(NULL);
  if (next_update_partition == NULL) {
    ESP_LOGE(TAG, "[E] update_from_file: Invalid OTA data partition or no eligible OTA app slot partition was found\n");
    return -1;
  }
  ESP_LOGI(TAG, "[I] update_from_file: Got OTA partition!\n");

  static esp_ota_handle_t handle = 0;
  ESP_LOGI(TAG, "[I] update_from_file: Begin OTA process\n");
  esp_err_t ota_begin_status = esp_ota_begin(next_update_partition, OTA_SIZE_UNKNOWN, &handle); // HERE THE APP CRASHES
```

and in the logs I get:

```
I (100) boot: Enabling RNG early entropy source...
I (106) boot: Partition Table:
I (110) boot: ## Label            Usage          Type ST Offset   Length
I (117) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (125) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (132) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (140) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (147) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (155) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (162) boot: End of partition table
I (167) boot: ota rollback check done
I (171) boot_comm: chip revision: 3, min. application chip revision: 0
I (178) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x066dc ( 26332) map
I (197) esp_image: segment 1: paddr=0x00026704 vaddr=0x3ffbdb60 size=0x018fc (  6396) load
I (200) esp_image: segment 2: paddr=0x00028008 vaddr=0x40080000 size=0x06dd0 ( 28112) load
0x40080000: _WindowOverflow4 at /home/tom/Documents/SLT/tecnocal/instacrop/fw/afr-test-ota/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1821

I (217) esp_image: segment 3: paddr=0x0002ede0 vaddr=0x00000000 size=0x01238 (  4664)
I (219) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x166d4 ( 91860) map
0x400d0020: _stext at ??:?

I (264) boot: Loaded app from partition at offset 0x20000
I (264) boot: Disabling RNG early entropy source...
I (264) cpu_start: Pro cpu up.
I (268) cpu_start: Application information:
I (272) cpu_start: Project name:     instacrop_app
I (278) cpu_start: App version:      1
I (282) cpu_start: Compile time:     Nov  6 2022 17:26:27
I (288) cpu_start: ELF file SHA256:  68f35072ce1355e0...
I (294) cpu_start: ESP-IDF:          v4.2.2-276-g4e4142ef8
I (301) cpu_start: Single core mode
I (305) heap_init: Initializing. RAM available for dynamic allocation:
I (312) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (318) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (324) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (330) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (337) heap_init: At 3FFC0A90 len 0001F570 (125 KiB): DRAM
I (343) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (349) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (355) heap_init: At 40086DD0 len 00019230 (100 KiB): IRAM
I (362) cpu_start: Pro cpu start user code
I (379) spi_flash: detected chip: generic
I (380) spi_flash: flash io: dio
I (380) cpu_start: Starting scheduler on PRO CPU.
I (391) afr-test-ota: [I] update_from_file: Got OTA partition!

I (391) afr-test-ota: [I] update_from_file: Begin OTA process

ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7880
load:0x40078000,len:13724
ho 0 tail 12 room 4
load:0x40080400,len:4584
0x40080400: _init at ??:?

entry 0x400806f4
I (58) boot: ESP-IDF v4.2.2-276-g4e4142ef8 2nd stage bootloader
I (58) boot: compile time 17:30:36
I (58) boot: chip revision: 3
I (62) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (69) boot.esp32: SPI Speed      : 40MHz
I (74) boot.esp32: SPI Mode       : DIO
I (79) boot.esp32: SPI Flash Size : 4MB
W (83) boot.esp32: PRO CPU has been reset by WDT.
W (88) boot.esp32: WDT reset info: PRO CPU PC=0x400837f3
0x400837f3: panic_print_char at /home/tom/Documents/SLT/tecnocal/instacrop/fw/afr-test-ota/amazon-freertos/vendors/espressif/esp-idf/components/esp_system/panic.c:69

W (94) boot.esp32: WDT reset info: APP CPU PC=0xcff5fa4
```

Thanks in advance.

AniruddhaKanhere, 2022-11-09:
Hello @diegoacuna,

I shall ask my colleagues @dachalco and @mahavirj to take a look at this questions.

Thanks for your patience,
Aniruddha

mahavirj, 2022-11-10:
@diegoacuna What happens if you try the OTA demo that comes with AFR? Do you see similar issue?

diegoacuna, 2022-11-15:
@mahavirj I can try that but is not what I need. We need to use the original OTA functions (because of the way we are fetching the firmware to upgrade). Using those functions is this not supported in amazon-freertos?